Nova-UTD/navigator #17: Transform map and localization information into a usable format

wheitman, 2021-10-29:


wheitman, 2021-11-02:
Hey team! Please [add your planning poker estimate](https://app.zenhub.com/workspaces/Nova-ZenHub-61760061bdbc0e001eae726d/issues/Nova-UTD/navigator/17?planning-poker) with ZenHub @crsz20 @EganJ @jim-works @NikhilNarvekar123

wheitman, 2021-11-02:
@jim-works would you mind adding some details about this in the Issue description?

jim-works, 2021-11-03:
Take in the lanelet map data and convert each lane and road boundary into a smaller length sequence of points such that creating the cubic spline through those points is a good estimate of the boundaries. I can fill in more details of the implementation once I look at the code.

wheitman, 2021-11-09:
@jim-works what's the progress on this? Could you use some extra help?

jim-works, 2021-11-12:
> @jim-works what's the progress on this? Could you use some extra help?

Nope, this is done except for unit tests. Turns out auto ware had a decent amount of code for this, which I copied and modified.

wheitman, 2021-11-19:
@jim-works Have these unit tests been written?

> > @jim-works what's the progress on this? Could you use some extra help?
> 
> Nope, this is done except for unit tests. Turns out auto ware had a decent amount of code for this, which I copied and modified.

Have these unit tests been written? @jim-works

joshmackwilliams, 2021-12-12:
Published in v1.1.0

closed 2021-12-12 by joshmackwilliams