Nova-UTD/navigator #54: Write wiki article describing outputs of ZED wrapper

wheitman, 2021-10-29:


wheitman, 2021-11-02:
Hey team! Please [add your planning poker estimate](https://app.zenhub.com/workspaces/Nova-ZenHub-61760061bdbc0e001eae726d/issues/Nova-UTD/navigator/54?planning-poker) with ZenHub @DragonflyRobotics @JamesMonCode @rae0924

rae0924, 2021-11-19:
[article](https://github.com/Nova-UTD/navigator/wiki/ZED-ROS2-Wrapper-Outputs)

wheitman, 2021-11-19:
Looks great! I assume that this document will be amended as our usage of the ZED wrapper expands?

We'll need to abstract this so that the downstream code (like the behavior planner) can't tell if the bounding boxes are coming from ZED or, say, the SVL simulator. I imagine we could just write a Python script to subscribe to either ZED Object messages or [LGSVL Detection3D](https://github.com/lgsvl/lgsvl_msgs/blob/master/msg/Detection3D.msg) messages, then republish as an abstracted voltron_msgs message that includes all necessary fields.

This abstraction would also allow code running outside of our [zed container](https://github.com/Nova-UTD/zed) to subscribe to object detection data, since the ZED SDK's custom Object message wouldn't need to be installed outside the container. Only our unified voltron_msgs package would be required.

You can go ahead and close this, @rae0924 . Good work.

closed 2021-12-12 by joshmackwilliams