Nova-UTD/navigator #145: Use GitHub Actions to test automatically

wheitman, 2021-10-31:


wheitman, 2021-10-31:
Please [add your planning poker estimate](https://app.zenhub.com/workspaces/Nova-ZenHub-61760061bdbc0e001eae726d/issues/Nova-UTD/navigator/145?planning-poker) with ZenHub @joshmackwilliams

joshmackwilliams, 2021-11-05:
I've noticed that some tests actually fail without extra setup. Part of this is actually my fault (the CAN interface is one of the failing nodes), but some of the ailing tests come from Autoware. Probably the first task associated with this will be to tweak these until all of the tests are passing. Alternatively, look into what these tests need in order to function correctly and work out how to replicate that in GH actions.

joshmackwilliams, 2021-11-11:
Looking at https://github.com/ros-tooling/action-ros-ci

I will probably create a separate repository to experiment with this to avoid running experiments on the main repo

joshmackwilliams, 2021-11-11:
Seems like we should use the Quad as a self-hosted runner - https://docs.github.com/en/actions/hosting-your-own-runners/about-self-hosted-runners

joshmackwilliams, 2021-11-17:
This is coming along quite nicely. We may not actually need to use the quad as a test runner, at least for the time being - each run will almost certainly take less than 10 minutes, and with 2000 free minutes per month, that gives us over 200 PRs per month. To put that in perspective, the total number of issues included in this first sprint is under 100. And from what I can tell so far, 10 minutes per run is a pessimistic estimate.

wheitman, 2021-11-17:
@joshmackwilliams very nice!

joshmackwilliams, 2021-11-17:
As always, this is proving to be more difficult than expected. I'm now testing in a branch on the main repo, and things are failing with obscure and unhelpful errors. 

Also, it's taking far longer than expected to build the workspace, likely because Actions only allows us to use a single thread. However, I think I've found a workaround - it should be possible to cache our build files, to take advantage of colcon's caching features (https://github.com/actions/cache), which would make builds happen much faster. 

I'm currently working on resolving the dependency issues. It's just slow going since the build happens in a totally ephemeral VM that I don't have direct access to.

joshmackwilliams, 2021-11-19:
Now using docker for the build and test, so I can run on my laptop (which runs faster than the GH hosted runner). I was running into issues with various tests failing, but that seems to be a quad-specific problem (probably related to our other network issues on the quad). I've started a test build, which might take up to an hour of so. We'll see how that goes once it finishes. If it's successful I think this issue can be merged (hopefully just in time for automatic testing to be included in our first release!).

joshmackwilliams, 2021-12-12:
Published in v1.1.0

closed 2021-12-12 by joshmackwilliams