Nova-UTD/navigator #146: Build Kinematic Motion Model

crsz20, 2021-11-02:


crsz20, 2021-11-02:
Define:

- [ ] Parameters
- [ ] Physical & Safety Constraints

wheitman, 2021-11-09:
@crsz20 what's the progress on this?

crsz20, 2021-11-10:
For the Kinematic Model, I recently started looking into the reference point for the inertial frame. Remember when we changed the point of origin for the PID controller to be the front axle of the wheels instead of the rear for Demo 1? So far, I'm convinced that these two are the same thing. Depending on the point of reference chosen, the controller design may vary.

I have also been looking at [this research paper](https://www.researchgate.net/publication/343428104_MPC-based_path_tracking_with_PID_speed_control_for_autonomous_vehicles). Looking at Figure 4 on page 7, I really like the structure of their controller. They happen to have the same goal we're looking for: Lateral MPC, coupled control for longitudinal PID, etc. However, there are two things I may have to look into:

1. They used _both_ a plant model and prediction model. I suspect that the prediction model is referring to the state variables. Most papers only reference the kinematic model, so I need to differentiate the two.
2. This is the first time I have come across a "DOF" model. How does it differ from Kinematic & Dynamic models?

wheitman, 2021-11-11:
@crsz20 I wonder if the DOF models, even the 8-DOF model, is more complicated than we need for Hail Bopp? For one thing, our vehicle has little to no suspension. Additionally, we have a very small speed range, about 0-23 mph or 0-10 m/s. It seems like[ the paper you cite](https://www.researchgate.net/publication/343428104_MPC-based_path_tracking_with_PID_speed_control_for_autonomous_vehicles) confirms your thinking on using lateral MPC and longitudinal PID, but do either require such a detailed model of the car?

A paper you included earlier, ([this one](https://ieeexplore.ieee.org/abstract/document/6957892)) *seems* to only require three parameters, but that paper isn't entirely clear to me.

@jruths mentioned just using a bike model. I know very little about any of this, but I'm simply wondering if there's a simpler solution than the one outlined in your paper above. That being said, if Fig 4 on page 7 seems like a good approach to you, then go for it. I would simply get Justin's advice first.

crsz20, 2021-11-11:
I definitely agree with that. The Kinematic model is great for low speed driving.

I did a bit of research into the DOF models. "DOF" refers to "Degrees of Freedom". Normally, there's 6 DOF for a rigid body in 3D motion (three rotational, three translational). The 8-DOF & 16-DOF comes from having additional models, such as the tire & suspension. This explains the additional complexity.

As for Figure 4, I was mostly referring to the layout & block diagram. Although 16-DOF supposedly brings the precision of a fully-drivable vehicle, I do agree that the Kinematic model should be sufficient for our implementation. 👍

jruths, 2021-11-11:
I think we will learn a lot just by implementing a simple model.  At the moment our initial focus should just be on getting a model that accounts for essentially position on the map and orientation.  Accounting for tipping and rolling of the vehicle will be good goals for the future - these are important, but less so when we're going slow.  So for now let's pick a model that just captures transverse movement.  I found [this article](https://dingyan89.medium.com/three-methods-of-vehicle-lateral-control-pure-pursuit-stanley-and-mpc-db8cc1d32081) online to be interesting and helpful.. you may have already found it.  I'm thinking that an approach similar to the one in Section 2.1.2 of [this paper](https://www.cs.cmu.edu/~motionplanning/reading/PlanningforDynamicVeh-1.pdf) could be a good way to start?  Don't get lost in the planning stuff they do after that section.

closed 2022-01-16 by wheitman