Nova-UTD/navigator #152: Create a basic safety manager

joshmackwilliams, 2021-11-04:
## Problem
We need to integrate most of our nodes with the safety manager. However, there is currently no concrete interface with which they can integrate, making development difficult. 

## Solution
We need a basic safety manager that will allow nodes to be tested. This will potentially involve development of a library for reliable communication with the safety manager. 

## Success condition
We have a safety manager that receives and logs events reliably and can reliably send commands back.

joshmackwilliams, 2021-11-04:
We can use a service (http://docs.ros.org/en/foxy/Tutorials/Services/Understanding-ROS2-Services.html) for reliable communication. Even if all we're sending back is an "ok" message, this model will guarantee that the desired target receives the message. For visualization and logging purposes, we can simply re-broadcast safety events from the safety manager itself. Alternatively, there may be a way to intercept and listen in on these calls - this will involve looking into how services are implemented. Each node also needs to run a service to receive commands from the safety manager as well.

joshmackwilliams, 2021-11-06:
Progress update: voltron_test_utils now has a TestServer and a TestClient, so I can now start developing the actual safety manager itself.

wheitman, 2021-11-08:
@joshmackwilliams any updates?

joshmackwilliams, 2021-11-08:
All that remains on this issue is to create a basic safety manager that logs messages when safety events are raised (I'm planning to leave the actual logic to another issue, which I will open shortly). This should only take a couple hours, but I'm pretty swamped with assignments and an upcoming exam atm. You can expect this to be done by next meeting for sure.

joshmackwilliams, 2021-11-08:
Also, turns out we will not need a special library (just use ROS servers and clients). I've put the srv files in voltron_msgs

joshmackwilliams, 2021-11-11:
Updated the technical doc to reflect what's currently been done as far as implementing the service interfaces

joshmackwilliams, 2021-11-18:
This has been reviewed and merged, so I'm closing it now

closed 2021-11-18 by joshmackwilliams