Nova-UTD/navigator #158: Use GitHub to enforce source control rules

joshmackwilliams, 2021-11-06:
GitHub has some mechanisms available for enforcing certain rules around source control - branch and merge using dev, don't commit directly to main or dev, and the like. These should be researched, and if they are helpful, applied to our repository to prevent mistakes in the future.

joshmackwilliams, 2021-11-11:
The reason that I'm starting this before the safety manager is finished is kinda complicated (basically boils down to the equals manager is still first priority, but I'm starting this in parallel

joshmackwilliams, 2021-11-11:
I've added rules to protect release, main, and dev branches. I'm not super familiar with these rules, so there is a nonzero likelihood that I've messed this up somehow. We'll need to watch how these rules play out as we go through the process of creating a release for the first time. 

@wheitman from a project planning/scrum/agile perspective, what should I do with this issue? It's more or less done until we test the new rules (by creating an actual release and merging it into main).

joshmackwilliams, 2021-11-11:
Also, to clarify what these rules are - Will and I are now required to review and merge all pull requests to dev, release, and merge branches. Two approvals are required for merges to main, while only one is required to merge into dev or a release branch. main and dev can't be deleted. You can no longer directly commit to main, dev, or release branches - modification has to be done via pull request. 

Obviously these are subject to modification, especially if they become cumbersome as we continue to develop.

wheitman, 2021-11-12:
@joshmackwilliams these look great. I can imagine that we'll modify these later on, as you say. I'll close this for now, since I've checked the settings and they appear to align with your intentions.  Later modifications to source control rules will not require reopening this Issue or creating a new one. It's just too meta, and it's something we can discuss personally.

closed 2021-11-12 by wheitman