Nova-UTD/navigator #161: Update lab digital signage

wheitman, 2021-11-11:


wheitman, 2021-11-19:
@Vion868 where do we stand here?

Vion868, 2021-11-24:
@wheitman Take a look at them, adding lots of photos of the car and going to add more of our weekly meetings! I have plenty more to add now that I have a good backlog of content

Vion868, 2021-11-24:
Need to add more transitions and I'd like to add more word slides, but I think that this is something that won't truly ever be done. I'm going to add more to it and change it all the time!

wheitman, 2021-11-24:
Looks great! I just made one minor edit to slide 2. Looking forward to seeing these slides evolve.

Vion868, 2021-11-26:
I'm planning now on adding interesting pics from each of our weekly meetings on there each week now (specifically if they look interesting...I.E. the more whiteboard drawings people do, the better!). My current backlog of photos means I can start rotating photos in and out at present without needing new ones, but I'll always be collecting more

wheitman, 2021-12-11:
I'll close this as soon as the timing on the slides is addressed @Vion868

Vion868, 2022-01-05:
@wheitman Issues with timing are done, they should all play automatically and on a loop! I will be adding more photos to these slides as long as we keep going of course, but for now, this issue is "done"

closed 2022-01-10 by wheitman