Nova-UTD/navigator #172: Design and implement hardware alarm system

joshmackwilliams, 2021-11-11:
We need a physical alarm system that receives an "all-clear" from the Jetson. Design and implement that alarm system. Note that the software support (controlling the alarm from the safety manager) is a separate issue!

joshmackwilliams, 2021-11-11:
@a-bainbridge I know you've done some work on this. If this issue is redundant, say so and I'll remove it. Otherwise, just let me know what progress has been made already.

wheitman, 2021-11-12:
Hmm, looks like #95 mainly, as well as
- #96 
- #97 
- #92 
- #91 
- #81 

@a-bainbridge ?

a-bainbridge, 2021-11-12:
Yes, this is definitely part of #95 and somewhat linked to #96, as in my solution would be incomplete without this. 

Our conversation the other night is pretty much all the exploration I've done on this issue so far, but I'll have a more fulsome approach developed within #96 when that's finished.

joshmackwilliams, 2021-11-12:
Great, I thought that was the case. I just directly transcribed my issues list from before the start of the sprint, so I'm not surprised there's overlap.