Nova-UTD/navigator #312: Add diagnostic data to all nodes

wheitman, 2023-01-19:
Each node should publish a [`DiagnosticStatus`](http://docs.ros.org/en/api/diagnostic_msgs/html/msg/DiagnosticStatus.html) message. A separate "watchdog" node should:
1. Aggregate these messages into a [`DiagnosticArray`](http://docs.ros.org/en/api/diagnostic_msgs/html/msg/DiagnosticArray.html) and publish this
2. Inspect each node's status and send an appropriate global status: "Ready," "Active," "Not Ready," "Critical Fault," etc.

stepan-antonyuk, 2023-01-25:
DiagnosticStatus will be publishing global status of a nodes.
DiagnosticArray publishes a status of all nodes based on their DiagnosticStatus.

stepan-antonyuk, 2023-01-27:
Lifecycle Node
Source: https://github.com/thehummingbird/robotics_demos/tree/main/lifecycle_node

pranavboyapati, 2025-08-04:
Add diagnostics data publishers to each node. Then, the guardian node will subscribe to each and determine an overall state