Nova-UTD/navigator #360: Map Manager refinement

danielv012, 2023-10-27:


jruths, 2023-10-30:
- This node currently publishes at about 3.5 sec for the driveable and junction grids.  Need to optimize this (I think doing more efficient lookup would do wonders).  While we will replace the driveable and possibly the junction with perception-based methods... having legacy support for the map-based approach is important to keep.
- Currently there seems to be some sort of race condition that leads to the map manager node crashing half the time upon launch?
- Needs to keep the "get_route" functionality as a service.  Currently the service does not reply back with the answer, but instead publishes a response... this should get fixed.  
- The get_route functionality only supports a start and end position and not intermediate points - also should get fixed.

jpahm, 2023-10-31:
I'll start working on this!

jpahm, 2024-09-17:
9/16/2024 Update:

    Status: Continuing work on previous task. Simultaneously looking into solving a pathing issue brought up by @akevinge
    Next Steps: Same as previous update.
    Projected Completion: Same as previous update.
    Update: Continuing work mentioned in previous update, but also looking into the pathing issue mentioned above. Pathing issue appears to be related to lane geometry and/or some sort of invalid coordinate frame.

jpahm, 2024-09-17:
> * This node currently publishes at about 3.5 sec for the driveable and junction grids.  Need to optimize this (I think doing more efficient lookup would do wonders).  While we will replace the driveable and possibly the junction with perception-based methods... having legacy support for the map-based approach is important to keep.
> 
>     * Currently there seems to be some sort of race condition that leads to the map manager node crashing half the time upon launch?
> 
>     * Needs to keep the "get_route" functionality as a service.  Currently the service does not reply back with the answer, but instead publishes a response... this should get fixed.  Will need to make corresponding updates to routing_monitor when you do this.
> 
>     * The get_route functionality only supports a start and end position and not intermediate points - also should get fixed.

Also, putting this here to keep track of the status of these 4 bullet point goals:

- [ ] Optimize publishing rate via lookup enhancement
- [ ] Replace driveable and (possibly) junction grids with perception-based methods, keeping support for map-based approach
- [x] Fix race conditions
- [x] Improve `get_route` service to reply appropriately and update routing_monitor accordingly
- [ ] Make `get_route` support intermediate points

jpahm, 2024-09-24:
### 9/23/2024 Update:

- **Status:** Continuing work from previous update.
- **Next Steps:** Same as previous update.
- **Projected Completion:** Still aiming for the 28th; may need to be pushed back to early October due to midterms and such.
- **Update:** Continuing work mentioned in previous update; still not sure on the cause of the pathing issue but diverting focus more towards the wrapper for now.

jpahm, 2024-10-01:
### 9/30/2024 Update:

- **Status:** Custom libOpenDRIVE changes are integrated into the wrapper
- **Next Steps:** Add final necessary wrapper functionality and finalize first version of python map manager
- **Projected Completion:** October 30th, aiming for sooner
- **Update:** Custom changes were integrated into the wrapper, now finishing other wrapper components and adding entire package to the navigator stack; will now finalize first draft of python map manager

jpahm, 2024-10-29:
### 10/28/2024 Update:

- **Status:** Finalizing Python conversion for C++ Boost objects
- **Next Steps:** Finalize first version of python map manager
- **Projected Completion:** Originally October 30, pushing back to early November due to multiple exams
- **Update:** Wrapper is essentially complete aside from needing to support Pythonizing some C++ Boost objects for geometric computations; should be able to quickly finish the first-draft Python map manager afterwards

jpahm, 2024-11-05:
### 11/4/2024 Update:

- **Status:** Finishing first Python map manager prototype
- **Next Steps:** Test and debug new map manager
- **Projected Completion:** November 10th
- **Update:** Ditched wrapping the existing C++ Boost RTree usage in favor of Pythonizing tree generation using the wrapper; now all geometry is handled via Shapely allowing for native Python support!

jpahm, 2024-11-12:
### 11/11/2024 Update:

- **Status:** Continuing work from last update
- **Next Steps:** Test and debug new map manager
- **Projected Completion:** November 15th
- **Update:** Have been very sick the past few days alongside having a large group project due on Sunday that delayed progress. Hoping to be finished by Friday.

jpahm, 2024-11-19:
### 11/18/2024 Update:

- **Status:** Continuing work from last update
- **Next Steps:** Continue debugging/testing new map manager, finalize first version
- **Projected Completion:** November 22nd
- **Update:** I have fully integrated the pyOpenDRIVE wrapper into the Navigator stack and have been testing builds using the new map manager; further additions, testing, and bug fixes are required before it is ready to be used.

jpahm, 2025-02-18:
### 2/17/2025 Update:

- **Status:** Implementing remaining functionality and implementing new routing (without C++ lemon lib)
- **Next Steps:** Implement routing in Python using either Djikstra or A*, then fill in remaining gaps
- **Projected Completion:** March 3rd
- **Update:** The wrapper functions properly and many of the C++ map manager's functionalities have been ported to Python and improved. The C++ pathing library (lemon) has no direct Python equivalent and must be replaced to fill in remaining gaps.

jpahm, 2025-02-24:
### 2/23/2025 Update:

- **Status:** Continuing work from last update
- **Next Steps:** Continue to implement routing, add support for intermediate point routing
- **Projected Completion:** March 3rd
- **Update:** Replacing lemon is non-trivial but doable, and I am making headway. Projected completion date may need to be pushed back further in order to accommodate porting the rest of the existing functionality.

jpahm, 2025-03-03:
### 3/2/2025 Update:

- **Status:** Continuing work from last update
- **Next Steps:** Same as last update
- **Projected Completion:** March 10
- **Update:** Continuing work from last update but I need more time.

jpahm, 2025-03-10:
### 3/9/2025 Update:

- **Status:** Finalizing re-implementation of existing functionality
- **Next Steps:** Finalize new routing system
- **Projected Completion:** March 17
- **Update:** Making more headway but there's still a decent chunk of work to be done.

jpahm, 2025-04-01:
### 3/31/2025 Update:

- **Status:** Continuing from previous update
- **Next Steps:** Same as before
- **Projected Completion:** April 14
- **Update:** Senior project and other commitments slowed my progress for a while, but I'm getting back on track with this now.

jpahm, 2025-04-21:
### 4/21/2025 Update:

- **Status:** Continuing from previous update
- **Next Steps:** Same as before
- **Projected Completion:** April 30
- **Update:** Was delayed further, but am in the final stretch now. Aiming to be completely done by May.