Nova-UTD/navigator #361: New path planner

danielv012, 2023-10-27:


jruths, 2023-10-27:
I have some ideas on this.

jruths, 2023-10-30:
- This will replace the current rtp node. The replacement node should be the path planner and produce a path that mostly follows the route, but deviates to avoid obstacles.  Based all on the summed cost maps.  
- I have in mind to have a grad student work on the optimization side of this and for someone to work to implement that optimization approach into the node.
- Still need to think if we need separate steering and speed cost maps?
- rtp right now also generates /vehicle/control, but that should be the responsibility of the controller (e.g., PID) to track the planned path as best as it can.

jruths, 2024-02-11:
This paper looks quite promising (and recent!):  https://arxiv.org/pdf/2401.13078.pdf#page2

This is implemented as part of the Nav2 ROS stack… we might want to look at some synergies here?