Nova-UTD/navigator #362: Localization / SLAM (KISS-ICP)

danielv012, 2023-10-27:


jruths, 2023-10-30:
This is a promising approach for ICP called KISS-ICP:  [https://github.com/PRBonn/kiss-icp](https://github.com/PRBonn/kiss-icp)

akevinge, 2024-09-10:
### 09/09/2024 Update:

- **Status:** Looking into previous work and debugging.
- **Next Steps:** Continue with the above until we can figure out both the U-turn failure and verify the accuracy of odometry. With both localization and lane detection, we should be able to create a naive path planning solution that can replace the route string. 
- **Projected Completion:** 10/09/2024
- **Update:** 
  - Attempting to replicate U-turn failure in Carla
  - Increase frequency of lidar in Carla
  - Reading through Ray's code and refactoring to understand
  - Poses from KISS ICP seem slightly off in Carla so looking into that
  - Worst case we switch off KISS ICP to IMU-compatible SLAM, but I think refining it is worth a try.

Attached is my monologue from Teams:
> Got a pretty interesting snap of what's happening. The turns within campus are pretty darn good (from what I can see visually), but it thought the U turn was a left turn. I don't think Ray realized we did a really hard U turn at the end of the route. I'm still digging into the paper, but my current line of thought is:
KISS ICP lacks global optimization and estimates poses incrementally (meaning drift can't be corrected w/o modifying KISS ICP itself). No amount* of PGO is going to save it from drift this hard. The U turn doesn't contain any unique features (IIRC it was just a tree line and houses on both sides)
Worth checking out LIO SLAM first (integrates with IMU + open source)
Worth checking out the other random SLAMs on the Kitti leaderboard
> [image]
> <br>
> <br>
> I'm still looking into KISS ICP, and I noticed /lidar is only ~5hz (os1's publish at 20hz), even on the Quad. I suspect low frame rate might be our culprit. You can see KIlSS ICP is registering the same lamp post in a different position here:
>[image]
> [image]
> [image]
> And when I decrease the convergence threshold (tighter convergence) and increase the iterations, it actually registers the U-turn (albeit with a good amount of drift).
>
> All this seems to suggest that there's too large of a difference between incoming point clouds = frequency is too low. Even the paper suggests 10-20hz.
> <br>
> <br>
> To follow up, the way KISS ICP is implemented in the primary open-source version:
Makes initial guess = last pose + (last delta)
ICP iteration, adjusting initial guess until:
>   - Convergence threshold is met
>   - Runs out of iterations
> 
> I think that the frequency of point clouds gets pretty sparse halfway during the U-turn, so it ends up using a partially corrected > guess - which would explain why it U-turns halfway, gives up, and keeps going left.
> <br>
> <br>
> [Here](https://github.com/Nova-UTD/vehicle_interface/blob/3c9cc174e1460429ac40c5c072ad344572c3790f/src/sensing/lidar/lidar/lidar_ouster_processing_node.py#L39-L44) you can see that only 1 /ouster/pointsmessage is kept in the buffer, which should probably be corrected to 20 as per [Choosing a good queue size](http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers#Choosing_a_good_queue_size).

> IMO localization definitely needs to include a safety node that slows the vehicle to a stop when GNSS odometry does not match up with pose from SLAM.

akevinge, 2024-09-17:
### 09/16/2024Update:

- **Status:** Trying to visualize and verify that localization is accurate / digging into map closure paper/code.
- **Next Steps:** Immediate action will be to test the accuracy of map closures via their pipeline. Moving forward with the assumption that failure at U-turns (and presumably sharp >90deg turns) is due to slow frequency of point clouds, verify that localization works and proceed from there (either to refine and integrate it, or to use something else entirely).
- **Projected Completion:** 10/16/2024
- **Update:** 
  - Digging into [map closures](https://github.com/PRBonn/MapClosures) it seems most of our localization logic is copied from the map closure pipeline (which provides us with theoretically accuracy odometry).
  - Finished refactoring original mapping script to be more readable (and for personal understanding)

akevinge, 2024-09-24:
### 09/23/2024 Update:

- **Status:** Trying to verify localization accuracy via `map_closure_pipeline`. 
- **Next Steps:** Again, verify that localization works and proceed from there (either to refine and integrate it, or to use something else entirely).
- **Projected Completion:** 10/16/2024 (very optimistic)
- **Update:** 
  - Working on a fork of [MapClosure](https://github.com/PRBonn/MapClosures/) to add ROS bag compatibility with their ground-truth evaluation pipeline. I think it'll be easier than writing our own, and I'm nearly done with it.

akevinge, 2024-10-01:
### 09/30/2024 Update:

- **Status:** Same as prior update but finished with modifying a [MapClosure](https://github.com/PRBonn/MapClosures/) fork to work with ROS bags.
- **Next Steps:** Same as previous update + record a bag with ground truth playing at the same frequency (same # of messages) as the point clouds.
- **Projected Completion:** 10/30/2024 (aiming for sooner)
- **Update:** 
  - Finished fork of [MapClosure](https://github.com/PRBonn/MapClosures/) to add ROS bag compatibility with their ground-truth evaluation pipeline. Ran into issue that ground truth needs to be same frequency as the point clouds (otherwise, it doesn't know which one to pick). It'll be easier for me to just record a proper bag than try modifying the fork any further.

akevinge, 2024-10-05:
ROS2 bag map closure evaluation: https://github.com/akevinge/MapClosures/commit/aa5b483. Just going to leave a process running on the quad and we'll see out it goes.