Nova-UTD/navigator #363: Automatic sensor calibration node

danielv012, 2023-10-27:


jruths, 2023-10-30:
There are cool ways of automatically calibrating the position and rotation of our sensors... either using just the normal environment or by using some key/easy-to-identify markers.

danielv012, 2024-02-02:
https://arxiv.org/pdf/2301.07279.pdf

VINDHYAkaushal, 2024-02-02:
https://github.com/tudelft-iv/multi_sensor_calibration

Min1it, 2025-02-18:
Status: Looking at other lab's implementations.
Next Steps: Collect ros bags of stop sign at different fixed locations away from the "origin" of hailbopp
Projected Completion: 4 weeks (mvp)
Update: Read relevant research. Looked at source code for other implementations. Ran the demo. Experimented with Ros bags.

Min1it, 2025-02-24:
### 02/24/2025 update:
**Status:** Preprocessing
**Next Steps:** Finish edge detection for LIDAR. Begin preprocessing on other topics.
**Projected Completion:** 3 weeks
**Update:** 
- Working on edge detection for LIDAR

Min1it, 2025-03-11:
**03/10/2025 Update:**
**Status: Preprocessing and Potential Roadblock**
**Next Steps:** Record a rosbag or start using Carla for testing. Potentially use/modify [image_projection_node](https://github.com/Nova-UTD/navigator/blob/dev/src/perception/segmentation/segmentation/image_projection_node.py).
Potentially communicate with people working on the "Camera-depth fusion" task.
**Projected Completion:** 3 weeks? (I feel like I need to record a new rosbag (existing sensors) to progress)
**Update:** 
Played with parameters a bit, managed to reduce pointcloud by factor of 10 while still clearly showing notable edges, etc. Difficult to say how much downsampling I'll be able to actually get away. As far as I can tell, the most recent non-carla bags (05/2024) in the quad only have data from the gnss sensor and lidar sensor (pointcloud and images), and don't have data from the Zed X cameras.

jruths, 2025-03-11:
Let's try starting with lidar and stereo camera point cloud registration.  This could be using several different methods.  Two classic ones are:
* ICP: https://en.wikipedia.org/wiki/Iterative_closest_point
* NDT: https://en.wikipedia.org/wiki/Normal_distributions_transform
It's highly possible newer methods exist too.

Min1it, 2025-04-01:
**03/31/2025 Update**
**Status:** ICP and Navigator Integration
**Next Steps**: Fully integrate into navigator
**Projected Completion**: 1-2 week (for code to be done)
**Update**: Got ICP working between two static point clouds using PCL. Not integrated into navigator currently. Current pipeline is simply agressively downsample each pointcloud then use ICP algorithm to synchronize them and get transforms. Extra considerations needed when registering more than 2 pointclouds. Depending on the reliability of our odometry, could potentially build a pointcloud of the world to allow calibration of sensors with no overlap with other sensors (for example, the rear facing stereo camera).

**Proposed Changes to navigator**: for each /lidar_source such as "/ouster/points", additional "/lidar_source/downsampled" topic will exist (differs from "/lidar/filtered" because of more aggressive downsampling and inclusion of the roof of the car). I propose "/lidar" be the synchronized point cloud topic, so that only the ground segmentation node will need minor changes.

**Future**: May be worth considering writing a node that takes some pointcloud and produce two pointclouds with differing levels of downsampling/processing for efficiency's sake (one pass). One will be used in "/lidar" while the other is used to find the transformations.

Min1it, 2025-04-07:
**04/07/2025 Update**
**Status:** ICP and Navigator Integration
**Next Steps:" Fully integrate into navigator
**Projected Completion:** 1 week
**Update:** No Update