Nova-UTD/navigator #365: Radar occupancy grid

danielv012, 2023-10-27:


jruths, 2023-10-30:
We'll be getting a new radar soon, so this should wait until that arrives.

ljmullings, 2024-10-01:
30/09/2024 Update:
Status: In the analysis phase 
Next Steps: Test publishing/receiving messages, learn how to run components, research best practices
Projected Completion: 15. October 2024
Update:
- None, ticket opened on 30/9/2024 😄

ljmullings, 2024-10-07:
07/10/2024 Update:
Status: Configuring a ROS node for point cloud simulations for 4d radar, Adding color coding for velocity, learning rosbag
Next Steps: Running Simulations with Point Cloud Data (defining how this simulation is going to work
Projected Completion: 18. October 2024
Update: 
- Learned occupancy grid basics
- Understanding format and structure of 4d radar's point cloud
- Ensured PCL is ROS Compatible
- Defining coordinate frames
- Investigated RVIZ (needs testing)
- Writing ROS node for color coding points and markers for velocity vectors
- check rosbag files/Gazebo for testing
Minor Block: hardware interfacing needs definition

ljmullings, 2024-10-29:
07/10/2024 Update:
Status: Confirmed Ethernet status, and learning Carla/Rviz
Next Steps: Running Simulations with Point Cloud Data/Test Data and checking subscribers, ensuring node is set up to detect messages on eth0
Projected Completion: 15. November 2024
Update:
- The configuration confirms that our radar sensors are set to communicate via Ethernet
- The configuration file doesn’t specify if the sensors support PoE directly. However, given that they connect over Ethernet, it’s likely feasible if they are PoE-compatible
- Each sensor is assigned a unique IP address with a common port (55555) for communication

ljmullings, 2024-11-19:
18/11/2024 Update:
Status: Planning DOGM algorithm
Next Steps: Testing PCL to Navigator and using CARLA pcl data to dogm.
Projected Completion: 15 December 2024
Update:
-  I want to shift from traditional occupancy grids to Dynamic Occupancy Grid Mapping (DOGM). Radar gives us velocity information, which means we can represent not just static occupancy but also the dynamics of the environment
- Each cell in the grid will store:
-- Probabilities for states like free space, static occupancy, dynamic occupancy, and unknown.
-- Velocity data, either as a mean or using particles to represent motion distributions.
- For the ISM, I’m developing a radar-specific approach that accounts for the unique challenges of radar data
- Radar has higher angular and elevation uncertainty than LiDAR, so I’m modifying the ISM to spread occupancy probabilities across cells based on this uncertainty
- Traditional ray tracing won’t work as well for radar because of angular uncertainty. I plan to replace it with a more probabilistic approach, like:
-- Probability Cone Models: Modeling radar measurements as cones of possible locations rather than precise rays.
-- Occupancy Probability Distribution: Distributing probabilities across cells in the measurement’s uncertainty area.
- Dynamic states will be updated based on observed velocities, and we’ll use state transition models to predict changes in occupancy and velocity over time.
-- I’m thinking about using Bayesian filters or particle filters for this.

ljmullings, 2025-02-18:
2/17/2025 Update:
Status: Roadblocked 
Next Steps: Testing the written and pushed container by subscribing from Navigator to Vehicle Interface
Projected Completion: The week the radar is completed with installation.