Nova-UTD/navigator #366: Dynamic occupancy grid prediction

danielv012, 2023-10-27:


jruths, 2023-10-27:
There is a "prednet" node (unused) that was working towards this. It may be useful or maybe not?

jruths, 2024-08-26:
Using the new 4D RADAR could be quite helpful here.

saiperam, 2025-02-18:
02/17/2025 Update:
Status: Got Started - So far I was able to research a paper on a dynamic occupancy grid and I was able to pull LIDAR data from a ROS bag into RVIZ for an occupancy grid. 
Next Steps: I plan on working with a physics based approach where I consider velocity of moving obstacles and for obstacles with velocity 0 (relative velocity as 0), there would be a static cell predicted. 
Projected Completion: Finish scraping data from LIDAR and other topics. 
Update: Effectively layed the groundwork with techniques for implementing a Dynamic Occupancy Grid and using RVIZ for a lidar topic. 

[image]

saiperam, 2025-02-24:
02/23/2025 Update:
Status: Was able to subscribe to the LIDAR Filtered Data published in RVIZ2! 
Next Steps: I plan on preprocessing the data coming from the /lidar/filtered topic to essentially analyze the positions of various moving and stationary objects. 
Projected Completion: Finish preprocessing and analyzing LIDAR data to demonstrate changes in an environment in real-time on a grid-based representation. 
Update: Effectively fixed the issue of LIDAR data from the ROS bag being published and seen as a topic with the "ros2 topic list" command. This helps in the next steps of preprocessing for implementing a physics based approach or a Kalman Filter approach.

saiperam, 2025-03-03:
03/02/2025 Update:
Status: I am currently researching existing models such as OpenPCDet for LiDAR Object Detection to identify dynamic vs static objects in LiDAR. Then, I'm researching how to implement models like a Motion Transformer or a Diffusion Model (essentially generates noisy data on top of LiDAR and the model is responsible for removing this noise and generating an accurate prediction of objects in the mere future). 
Next Steps: I plan on refining a pipeline of models to implement a grid that not only shows where points in the future will be but rather indicate the future movement of obstacles in real-time! 
Projected Completion: Finish model implementation to track objects in LiDAR as well as their future movements! 
Update: Working on cloning a repo of models that are responsible in identifying objects in LiDAR data as well as predicting their future movements within a grid.

jruths, 2025-03-03:
Sounds great!  We do have a lidar-based object detection node in the stack already.  Is there any way you can use the topics from that node so you don't have to repeat that part?

saiperam, 2025-03-31:
03/31/2025 Update:
Status: I was successfully able to run our existing Lidar 3D Object Detection model! I got the static occupancy grids to also appear in real time when the ROS bag is playing! 
Next Steps: I am working on preprocessing the data and saving them into timestamps to train a Scene Transformer in predicting the future movements of various agents in an environment, grid-cell representation. 
Projected Completion: Preprocess the information retrieved from the topics into timestamps for feeding into input for the Scene Transformer model. This transformer model will then produce occupancy grids in future timestamps! 
Update: I successfully got the RVIZ2 Occupancy Grids working! 

[image]

saiperam, 2025-04-07:
04/7/2025 Update:
Status: I successfully stored timestamped data of static occupancy grids as well as color coded these grids: black for cells that have remained unchanged (previously free and still free, previously occupied and still occupied) and red for cells that have changed (previously free and now occupied or previously occupied and now free). 
Next Steps: I am working on preprocessing this data to train a Scene Transformer model in making predictions of what cells will be changed in the future by taking input 5-10 occupancy grids in the past. 
Projected Completion: Pull a Scene Transformer model to train it on the occupancy grids to make future predictions based on which cells have changed and which have not! 
Update: I successfully stored the occupancy grids as timestamped data as well as color coded it based on changes that were made within each cell! 

[image]

[image]

saiperam, 2025-04-14:
04/13/2025 Update:
Status: I successfully trained a Long Short Term Memory (LSTM) Network on this timestamped data of static occupancy grids to take input 10 frames and forecast a new occupancy grid for the next timestamp! 
Next Steps: I am evaluating the performance of this network by ensuring that the predictions align with the actual occupancy grids. I also plan on adding the LiDAR Object Detection model in conjunction with this output to color cells that are dynamic objects in real time (cars, pedestrians, trucks). 
Projected Completion: I plan on integrating this model into ROS2 to publish topics visualizable in RVIZ2 for real-time forecasting of not just the occupancy grid, but also LiDAR Object Detection! 
Update: I have trained an LSTM to make predictions of future occupancy grids by taking input 10 grids! 

[image]

saiperam, 2025-04-29:
04/28/2025 Update:
Status: I successfully integrated the trained LSTM model to make predictions 5 seconds into the future as a ROS2 topic! 
Next Steps: I plan on visualizing it in RVIZ2 to see if the cells are properly being predicted in comparison with the actual occupancy grid running in RVIZ2. 
Projected Completion: I plan on fixing the errors I am getting with running the LiDAR Object detection node to classify objects in the occupancy grid. 
Update: I have trained an LSTM to make predictions of future occupancy grids 5 seconds + publish them as topics! 

[image]

[image]