Nova-UTD/navigator #372: Drivable surface segmentation (image)

danielv012, 2023-10-27:


jruths, 2024-08-13:
Currently the costmap for driveable surface is provided by the map manager.  While this is a reasonable place to get a driveable surface map, it tends to be slow and creates some significant lag on the realtime aggregated costmap.  It also puts possibly unnecessary reliance on the map.  The goal here is to use camera and possibly lidar to identify the surfaces around the vehicle that are driveable (road) so it can be used to create a costmap for road/not-road.

jruths, 2024-08-28:
There currently exists a image_segmentation_node.py that may do some of this or provide a starting point?  I don't think we currently use this node, so it is untested (or at least hasn't been tested in the past year).  If it works and one of the "classes" it produces is driveable road, then creating the costmap would largely be about mapping the image into a costmap (projecting the image into 3D space... which is related to another task: #370 ).

This node is located at src/perception/segmentation/image_segmentation_node.py

saishravanm, 2024-09-10:
### 09/09/2024 Update:

-Status: Researching how to use a Voxel Grid to convert the 2D occupancy grid (returned by the existing image_segmentation_node) into a 3D costmap. 
-Next Steps: Work on creating a Voxel Grid. 
-Projected Completion: 9/16/24
-Update:  Commented and understood the existing image_segmentation_node and image_projection_node.

saishravanm, 2024-09-17:
### 09/16/2024 Update:

- **Status:** Doing a deep dive into MMS documentation for the best methods of 2D image-segmentation
- **Next Steps:** Get a demo of CARLA working (the car moving along the city) and make any dataset changes(or create a new one entirely) for the node.  
- **Projected Completion:** 10/4/24
- **Update:**  
-Last week, David and I separated the objectives for the image_segmentation_node and the image_projection_node. I will be working on the 2D image segmentation portion which will return a 2D occupancy grid and be fed into the image_projection (2D to 3D) service which David will be working on. 
-I was able to get a prebuilt demo of the MMSegmentation framework (based on PyTorch) working (had some small issues with matching versions). Using that I was able to better understand the current image_segmentation_node and the important parts of how 2D image segmentation works. The existing node currently uses a pre-trained dataset. Still need to test it out with CARLA to see if it is sufficient, if not, must be tweaked. 

- **Resources Used:**
-MMSegmentation Documentation: https://github.com/open-mmlab/mmsegmentation/tree/main
-Fixed MMSegmentation Demo (with the correct versions): https://colab.research.google.com/drive/1l0fpP8NCxlU8QAzD4KfR3dmyzlfAgEEC?usp=sharing

saishravanm, 2024-09-24:
### 09/23/2024 Update:

- **Status:** Rewriting existing image_segmentation node with the latest versions of MMSegmentation, as well as rewriting manual_control node with Pygame instead of ncurses. 
- **Next Steps:** Test current image segmentation logic in CARLA, using the keyboard controls
- **Projected Completion:** 10/6/24
- **Update:**  
-My current approach in terms of finishing this drivable surfaces task as a whole, is to work off of the current image_segmentation node and edit the model as necessary. In order to do that, I need to get a working keyboard controller for CARLA up and running so I can test how well the current code works in the simulation. 
-I spent most of my time trying to work through the CARLA Launch file environment and fixing package related bugs in both of these nodes. (The manual_control node wouldn't launch at all because the word "pedal" was misspelled throughout the entire file)
-The manual_control node provides a terminal based GUI(using ncurses) for vehicle control, however this requires a separate terminal just for controlling. I'm currently working on utilizing pygame instead, which should allow for vehicle control in the same terminal. After that, I plan on getting a GUI panel in CARLA to directly interface with the controls. 
-The image_segmentation node uses an older version of MMSegmentation, which uses a slightly different approach to how functions are called (especially how pre-existing models are used). As a result, the node doesn't work as of now. I'm still going through the documentation and making the necessary fixes so I don't have to rewrite all the logic.

saishravanm, 2024-10-01:
### 09/30/2024 Update:

- **Status:** Testing image segmentation model parameters using the latest MMSegmentation tools.
- **Next Steps:** Fine-tune the model and integrate it as a viewable "layer" in CARLA. Resolve issues with the keyboard_control node logic.
- **Projected Completion:** 10/12/24
- **Progress Update:**
  - I successfully began implementing a new `keyboard_controller` node in CARLA using Pygame. However, due to multiple existing nodes (specifically the routing nodes) accessing the vehicle's position topics, I’m facing issues getting the vehicle to move properly. This requires further troubleshooting.
  - For image segmentation, I’ve updated the code to the latest OpenMMS version and successfully ran the node in CARLA without errors. However, debugging is still needed as the node currently doesn’t output easily interpretable results. I'm now working with the `SegLocalVisualizer` tool to test and adjust the current coloring scheme on test images.

saishravanm, 2024-11-04:
### 11/04/2024 Update:

- **Status:** Tweaking Image Segmentation model parameters. 
- **Next Steps:** Get LiDAR point cloud info and segmented image data and create segmented point cloud.
- **Projected Completion:** 11/10/24
- **Progress Update:**
  - On Friday (11/1/24) I was able to complete the keyboard_controller node which allows for Hero to be controlled via WASD and X in CARLA. It's sort of sensitive to input (takes exponential time to reach top throttle) so ensure that you press and hold the keys as seen fit. The node is in the "keyboard_controller" branch. 
  - For image segmentation, I decided to completely start afresh and rewrite everything in terms of Facebook's SAM 1.0 (Segment Anything Model) and was able to get a demo of it in action in CARLA. It took a while but I think SAM is the most optimal way to go because of easy it is to implement (there is a 2.0 that exists but I didn't see beforehand but shouldn't be too hard to port it over). 
  - 45s Demonstration Video: 
[image]
  - I did some research on how to generate the 2D occupancy grid after completing the segmentation stuff. I decided to use this paper: https://ieeexplore.ieee.org/document/9605417 (Occupancy Grid Map Estimation Based on Visual SLAM and Ground Segmentation). 
  
I adapted the diagram that was on it to include the LiDAR point cloud information as well. 
https://app.diagrams.net/#G1qH1k9xMugIRHEagD866GFj8gvjelf8OF#%7B%22pageId%22%3A%22JFvVPnVwIeuTqtvLyU7o%22%7D

saishravanm, 2024-11-12:
### 11/11/2024 Update:

- **Status:** Researching how to convert segmented 2D image into occupancy grid
- **Next Steps:** Utilize RTAB-MAP to create either a stereo map with images, or occupancy grid with segmented information laid on top
- **Projected Completion:** Complete functionality by 11/18
- **Progress Update:**
  - I was able to resolve the "issues" in the keyboard_controller_node and updated the PQ. 
  - I ported the current image_segmentation_node from SAM 1.0 to SAM 2.0 as the latter makes it easier to gather information about the segmented portions of an image. 
[image]
 - I am currently researching the best way to create the 2D occupancy grid using the segmented information and so far the best option is to utilize RTAB-MAP (an RGB-D, Stereo and Lidar Graph-Based SLAM approach based on an incremental appearance-based loop closure detector). The issue is that currently, the segmentation and projection functionality are split between my node and David's node, whereas RTAB-MAP is a ROS2 node on its own which combines both functionalities. 
 - RTAB-MAP Demo videos: https://www.youtube.com/watch?v=qpTS7kg9J3A
 - https://github.com/introlab/rtabmap/wiki/Stereo-mapping

saishravanm, 2025-02-10:
### 02/09/2025 Update:

- **Status:** Researching ways to tweak SAM 2.0 segmentation model. 
- **Next Steps:** General bug testing, tweak model parameters, and ZED compatibility to make PR.
- **Projected Completion:** Make PR by 2/19.
- **Progress Update:**
  - I was able to successfully generate an occupancy grid of the drivable surface based on segmented image data and depth map taken from the latest rosbags. 
  - This was done by creating a "drivable surface pipeline" (3 nodes):
    - image_segmentation: takes images from **/ouster/signal_image**, applies multiple segmentation masks(chose the one with the highest confidence score), converts mask into binary mask (255(white): drivable, 0(black): non-drivable), then publishes to topic **/segmentation_mask**
    - depth_perception: takes images from **/ouster/range_image** , then converts the received cv2 images to a 32 bit single channel grayscale depth map, and then publishes to topic **/processed_depth**
    - occupancy_grid: creates ROS2 occupancy grid, uses segmented mask to determine whether grid point is drivable or not, uses depth map to determine whether grid point is occupied/free, and publishes grid to topic **/occupancy_grid**
 - I haven't tweaked much of the model parameters so the segmentation is still shaky, will focus on that for the next goal
 - RTAB-MAP Demo videos: https://www.youtube.com/watch?v=qpTS7kg9J3A
 - https://github.com/introlab/rtabmap/wiki/Stereo-mapping

saishravanm, 2025-02-25:
### 02/24/2025 Update:

- **Status:** Fine tuning SAM 2.0 Model
- **Next Steps:** Implement Mapillary Vistas (Street-level images dataset) as ground truth, implement Track Mode for continuous detection across frames, and Kalman filter for noise handling
- **Projected Completion:** Complete Next Steps by March 4th
- **Progress Update:**
  - The segmentation model doesn't do so well using the default dataset that it came with, so the first step is to train the model on Mapillary Vistas, which is a dataset used to understand street scenes all over the world. By doing so, the model should develop a baseline for classifying roads versus other objects in the foreground
  - The second step is to implement Track Mode, which adjusts the mask applied to Frame *n* to Frame *n+1* and so forth. Doing so should allow some level of consistency between frames (which prevents warping) 
  - The final step is to implement Kalman filter, which is used to interpret the "uncertainty" of the system (in this case the uncertainty of the mask) and provides a probabilistic approach to update the system's state estimate based on prior knowledge and new measurements. By doing so, we should be able to achieve the most "optimal" or "accurate" mask taking into account the current state, noise (specifically between certain frames), and any noise present.
  - Resources Used:
      - Mapillary Vistas Dataset Link: https://www.mapillary.com/dataset/vistas
      - Kalman Filter Tutorial: https://www.scaler.com/topics/kalman-filter-in-computer-vision/

saishravanm, 2025-03-04:
### 03/03/2025 Update:

- **Status:** Retraining SAM 2.0 Model on Mapillary Vistas
- **Next Steps:** Retrain model and then test for performance
- **Projected Completion:** March 7th
- **Progress Update:**
  - Mapillary Vistas, a large dataset containing segmented street-view images, seems like a good ground truth to retrain the SAM 2.0 model on (which is currently having the default weights).
  - I wrote a script turning all the segmented masks to binary masks (script took overnight to run) and now I'm in the process of figuring out how to retrain a SAM 2.0 PyTorch model.
  - I have created the PyTorch Dataset containing the images and label masks, and now I'm trying to debug the forward propagation functions in SAM 2.0
  - Since the Quad didn't have the 32 GB space necessary to download Mapillary Vistas, I have been doing everything on my local machine, which is taking quite a while to compile and run things. Is it possible to clear up some space on the Quad? 
  - Example Mapillary Vista Image Masks:
  - [image]
  - [image]
  - [image]

saishravanm, 2025-03-24:
### 03/24/2025 Update:

- **Status:** Currently working on correcting the occupancy grid generation process.
- **Next Steps:**
  - **Occupancy Grid Generation**
    - Implement correct 3D point projection using depth + camera intrinsics/extrinsics
    - Publish the completed grid to `/occupancy_grid`
  - **SAM 2.0 Fine-Tuning**
    - Re-train SAM 2.0 using Mapillary Vistas as ground truth
    - Implement Track Mode
- **Projected Completion:** 03/29/2025
- **Update:**  
  - While out of town for the last 2 weeks, I didn't have the best internet connection to work comfortably and reliably in the Quad, so I used the time to revisit and review my original pipeline for converting segmented images into an occupancy grid using depth maps.
  - I initially implemented the occupancy grid incorrectly by trying to directly use the segmented image output without transforming it into 3D coordinates and binning it spatially according to a real-world grid. I’ve now clarified the correct pipeline:

   1. Use camera intrinsics (focal lengths and principal point) with the depth and pixel data to project segmented pixels into 3D
   2. Use ROS2 TF to transform points from the camera frame (`zed2_left_camera_optical_frame`) to a common vehicle frame (`zed2_camera_center`)
   3. Bin the transformed points using a fixed-resolution top-down occupancy grid
   4. Format and publish the grid

  - While away, I had already completed processing the Mapillary Vistas' segmented masks (separates objects into different colors) into binary masks (road is white and environment is black) on my home computer. I just need to transfer those to the Quad to continue retraining the model.
  - Note: The frames from the latest rosbag "rosbag2_2024_05_10-loops-neighborhood/rosbag2_2024_05_10-18_52_20_0.db3" only has information on one of the cameras on the car (presumably the only one being used at the time of recording). In order to calibrate the remaining 3 camera frames (the physical location and orientation of the cameras) onto the common vehicle frame (the physical location and orientation of the vehicle), I'd need a rosbag with the remaining cameras working. 

[image]

saishravanm, 2025-04-01:
### 03/31/2025 Update:

- **Status:** Fine-tuning SAM 2.0 using Cityscapes binary road masks; currently resolving a decoder input shape mismatch.
- **Next Steps:**
  - Fix final `RuntimeError` from `dense_prompt_embeddings` mismatch.
  - Confirm shape alignment with debug printouts before decoder call.
  - Run training loop once input shapes work.
- **Projected Completion:** ???
- **Update:**  
  - After returning back home, I completed preprocessing the Cityscapes dataset. I stopped using Mapillary Vistas as I kept running into errors trying to get the sampler to work
  - Wrote a custom PyTorch dataset class to load paired images and masks and verified that they align.
  - Integrated SAM 2.0 model with a dummy sparse prompt and GT mask as the dense prompt.
  - wrote script to perform supervised fine-tuning using SAM’s decoder.
  - However, I hit a blocker with `dense_prompt_embeddings` not matching the expected shape (`[B, 256, H_feat, W_feat]`) inside the SAM decoder.
  - I attempted multiple fixes including resizing and expanding the GT masks but still receive a shape mismatch error from the model.

pranavboyapati, 2025-04-19:
### 04/18/2025 Update:

- **Status:** Reviewed resources related to this task
- **Next Steps:** Prepare a dataset for retraining the image segmentation node
- **Projected Completion:** Monday, April 28
- **Update:**  I reviewed all of the documentation Sai provided for this task and looked over the existing code. I also reviewed some articles about image segmentation and the tools that we are using for this task. The node has a couple of issues in its current state, so the first step to rectifying these is to retrain the SAM2.0 model. So, I will prepare a dataset (most likely using Mapillary Vistas) for retraining the model.

pranavboyapati, 2025-04-25:
### 04/25/2025 Update:

- **Status:** Built a new dataset for retraining the SAM2.0 model
- **Next Steps:** Retrain the SAM2.0 model and test it for accuracy
- **Projected Completion:** Monday, May 5
- **Update:**  I built a dataset of 3000 images from Mapillary Vistas and their corresponding binary masks. These will be used to retrain the SAM2.0 model for the image segmentation node and resolve the issues with it. This dataset is much larger than the initial dataset Sai used, so it should be better for training the model and making it more accurate. I will train this model and test it to ensure accuracy. A backup plan in case retraining the model doesn't help is to switch to a different model like YOLO.

closed 2025-07-01 by jruths