Nova-UTD/navigator #375: Road signs classifier

danielv012, 2023-10-27:


jruths, 2024-08-27:
This should take in camera data and output bounding boxes that contain the location and classification of the sign.

VINDHYAkaushal, 2024-09-16:
**Status**- On pause till traffic light detector works

pranavboyapati, 2024-10-15:
### 10/14/2024 Update:

- **Status:** Task just picked up by me
- **Next Steps:** Make a plan for completion and also make a flowchart showing progression of logic
- **Projected Completion:** Monday, October 21
- **Update:**  I just picked up this task, so I will begin by planning out how to complete the task. I will make a plan and also a logic flowchart for this task by next week.

pranavboyapati, 2024-10-22:
### 10/21/2024 Update:

- **Status:** Made a plan for completion and flowchart outlining logic [(linked here)](https://cometmail-my.sharepoint.com/:i:/g/personal/pkb230000_utdallas_edu/EXiSxGFS619AtiwxPUgw8AoBxuqiVqzXCrqJeoxCz7qqKg?e=ro4tbQ)
- **Next Steps:** Look for a dataset for training the model and existing repos that will help with this
- **Projected Completion:** Monday, October 28
- **Update:**  I planned out how I will complete this task, and I also made a flowchart outlining the logic that will be followed by the road signs classifier. My next steps are to find a dataset which I will train the model with and other resources that will help with this. I will finish this within one week.

pranavboyapati, 2024-10-29:
### 10/28/2024 Update:

- **Status:** Found a preexisting model that works for our case. Also found a dataset to train a model if we choose that instead.
- **Next Steps:** Assuming we will use the preexisting model, the next steps are to write code for preprocessing the image, running it through the model, and publishing the classification.
- **Projected Completion:** Monday, November 4
- **Update:**  I found a preexisting model that classifies US road signs with 94% accuracy (according to where the model was made available). I conducted additional testing on this model to ensure it meets our needs and it does. I feel the best course of action is to use this preexisting model because it does the job perfectly. In that case, the next steps are to write the code for running images collected by cameras on the car through the model and publishing the classification as a message. If we choose to build our own model instead of using the preexisting one, I found a dataset of road signs images and tutorial-like resources I can use to build our own model.

pranavboyapati, 2024-11-05:
### 11/04/2024 Update:

- **Status:** All coding for the road signs classifier is finished
- **Next Steps:** Test this out on CARLA
- **Projected Completion:** Monday, November 11
- **Update:**  I finished all of the coding for the road signs classifier. The following logical progression is followed: take in camera images, preprocess the image, make an API call to the model, publish the classification and position information. The next step is to conduct thorough testing using CARLA, which I expect will be finished in one week.

pranavboyapati, 2024-11-11:
### 11/11/2024 Update:

- **Status:** Opened pull request

closed 2024-11-13 by pranavboyapati