Nova-UTD/navigator #376: Intersection manager

danielv012, 2023-10-27:


pranavboyapati, 2024-09-09:
### 09/09/2024 Update:

- **Status:** Created a detailed plan for completing task
- **Next Steps:** Create functionality to classify the type of intersection (roundabout, stop sign, traffic light)
- **Projected Completion:** Monday, September 16
- **Update:**  I ran Navigator and Carla on my computer to fully understand how these systems work and understand the current capabilities of the Navigator stack. I also created a detailed plan for how I can complete the intersection manager task, noting down details of what all behavior is needed in different scenarios and identifying potential dependencies. When creating functionality for classifying the type of intersection, I will need the traffic light detector and road signs classifier, neither of which are completed yet, so I or someone else will need to work on these. The map manager may also be needed, but this is already finished, so there won't be any problems with using that.

pranavboyapati, 2024-09-16:
### 09/16/2024 Update:

- **Status:** Finished as much as possible without the road signs classifier for determining the intersection type
- **Next Steps:** Create functionality to check if the car is in the correct lane for the direction it wants to go through the intersection
- **Projected Completion:** Monday, September 23
- **Update:** I created a package for the intersection manager and began writing the code for this. Currently, it is possible to detect if an intersection is controlled by a traffic light. I was hoping that I could also be done with the code for classifying stop-sign-controlled intersections and roundabouts, but was unable to do so due to the road signs classifier not being completed/close to completed. I will go back and add in the part for determining stop-sign-controlled intersections and roundabouts once the road signs classifier has more progress made. My next step is to create the functionality for checking if the car is in the correct lane for going through the intersection (e.g. right only lane for right turns, left only lane for left turns, etc.).

pranavboyapati, 2024-09-24:
### 09/23/2024 Update:

- **Status:** Finished as much as possible for checking if the car is in the correct lane for going through the intersection without having the lane detection or lane change nodes available
- **Next Steps:** Develop functionality for safely navigating a traffic light controlled intersection
- **Projected Completion:** Monday, October 7
- **Update:**  Currently, the function for determining if the car is in the correct lane for going through the intersection is in a partially completed state. I did as much as was possible not having the lane detection or lane change nodes available, with placeholder comments where these nodes will be used. I will go back and replace these comments once the lane detection and lane change nodes are completed. My next step is to develop the functionality for driving through a traffic light controlled intersection, with the logic followed outlined on the flowchart in my previous update. I believe all the nodes that are needed for this part are already completed, so there shouldn't be any issues there. I have planned two weeks to complete this part, but I may be able to finish in less time than that.

pranavboyapati, 2024-10-01:
### 09/30/2024 Update:

- **Status:** Finished functionality for traffic light controlled intersections
- **Next Steps:** Develop functionality for navigating stop sign controlled intersections
- **Projected Completion:** Monday, October 14
- **Update:**  The functionality for safely driving through a traffic light controlled intersection is fully completed. Unlike in past stages of this task, there weren't any dependencies that stopped this from being fully completed. The next step is to develop the functionality for navigating a stop sign controlled intersection. I do not expect this to be held up because a dependency is not completed, but I may need the road signs classifier, and that dependency will cause this step to be held up and not be fully completed. I am planning two weeks to complete this step, but it is highly likely that it will be completed in less time than that.

pranavboyapati, 2024-10-08:
### 10/07/2024 Update:

- **Status:** Finished as much functionality as possible for stop sign controlled intersections given some nodes needed are not done yet
- **Next Steps:** Develop functionality for navigating through roundabouts
- **Projected Completion:** Monday, October 21
- **Update:**  The functionality for driving through a stop sign controlled intersection is mostly completed, with the only parts not done yet being those that require a node that is still in progress. The road signs classifier and the lane type detection nodes are needed for fully completing this section of the intersection manager, but they are not completed yet, so I added placeholder comments where features from those nodes will go. The next step is to develop functionality for driving through a roundabout, which may also require nodes that are not completed yet but I am not sure of this at the moment. I am planning two weeks to complete this part, but it may be completed in less time, like has happened with other parts of this task.

pranavboyapati, 2024-10-15:
### 10/14/2024 Update:

- **Status:** Intersection Manager finished as much as possible at this time
- **Next Steps:** Move on to other task until dependencies are ready
- **Projected Completion:** ?
- **Update:**  The intersection manager has been finished as much as possible given the currently available nodes. Once the rest of the dependencies are finished, the placeholder comments can be replaced with the correct code. This task will be moved to the to-do section and will be resumed once dependencies are finished.

pranavboyapati, 2025-02-16:
### 02/17/2025 Update:

- **Status:** Just resuming this task now
- **Next Steps:** Add all the dependencies to the ROS node
- **Projected Completion:** Monday, March 3
- **Update:**  All of the dependencies for this task are now completed, so I can now resume working on this. I will work on adding in all of the dependencies that are now ready. Once the dependencies are added in, I will finish the other remaining parts of the node. I am expecting it to take about 2 weeks to finish the node and verify that the logic in the flowchart is correctly followed. Then, I will test on Carla/ROS bags.

pranavboyapati, 2025-02-23:
### 02/23/2025 Update:

- **Status:** Finished making the ROS node
- **Next Steps:** Test on Carla
- **Projected Completion:** Monday, March 3
- **Update:**  I added in all of the dependencies to the node, so it is now fully complete. I will use this week to test out the node on Carla and work out any issues that come up.

pranavboyapati, 2025-02-28:
### 02/27/2025 Update:

- **Status:** Finished this task
- **Update:**  I finished testing this node on Carla and it works properly, so I opened a pull request.

closed 2025-06-30 by jruths