Nova-UTD/navigator #378: Pedestrian skeleton

danielv012, 2023-10-27:


jasonl112, 2025-01-27:
### 01/27/2025 Update:

- **Status:** Just picked up by me! I am leaving Hand Gesture Recognition until after I finish this task.
- **Next Steps:** Dive deeper into the CV models and implement/test them out for pose detection.
- **Update:**  This week, I looked into some CV models for pose detection, such as MediaPipe, Detectron, and OpenPose!

jasonl112, 2025-02-11:
### 02/10/2025 Update:

- **Status:** Almost Done with Pedestrian Skeleton!
- **Next Steps:** Finish integrating, testing, and making a PR!
- **Projected Completion:** 02/17
- **Update:**  I began integrating into my own ROS node, but I didn't get the chance to finish because I have been busy with doing interviews for ACM!

jasonl112, 2025-02-18:
### 02/17/2025 Update:

- **Status:** Almost at the finish line with Pedestrian Skeleton!
- **Next Steps:** Finish integrating, writing documentation, and making a PR!
- **Projected Completion:** 02/19
- **Update:**  My main blocker is DOCKER . My Docker image/containers basically blew up, and I spent most of my time this week trying to fix it.  I fixed it and now I am testing my pedestrian_skeleton node with more rosbags.

jasonl112, 2025-02-20:
### 02/19/2025 Update:

- **Status:** Pedestrian Skeleton Done!
- **Update:**  I made a PR and waiting for review!

closed 2025-06-27 by pranavboyapati