Nova-UTD/navigator #380: Pedestrian gesture classifier

danielv012, 2023-10-27:


jasonl112, 2024-11-12:
### 11/11/11 Update:

- **Status:**  In Progress – Currently researching datasets; have identified several collections focused on hand gestures, often used for sign language recognition.
- **Next Steps:** Begin training the model to recognize two specific gestures: “stop” (open palm) and “move forward.” Planning to train the model with a custom or premade dataset.
- **Projected Completion:** Monday, November 18, 2024
- **Update:**  The model successfully detects cars, people, and other general objects. Aiming to complete gesture training within one week.

jasonl112, 2024-11-19:
### 11/18/2024 Update:

- **Status:** In Progress – Currently fine-tuning the model to recognize hand gestures from pedestrians at a distance.
- **Next Steps:** After fine-tuning, I will test this out on CARLA and open a PR.
- **Projected Completion:** Monday, December 2nd, 2024
- **Update:**  The model successfully detects hand gestures and points of the hands when the person is close to the camera.

jasonl112, 2024-12-03:
### 12/02/2024 Update:

- **Status:** In Progress - Researching methods to improve the model’s ability to detect hands and classify gestures, even in challenging scenarios such as distant, blurry, or briefly visible hands.
- **Next Steps:** Select one, implement it, and check if it works!
- **Update:**  Pivoted to a hybrid approach by leveraging YOLOv8 for initial hand detection, followed by OpenCV and Mediapipe for detailed landmark analysis to detect hand gestures in challenging scenarios. The detection models works on webcam and high quality pictures, but not quite for blurry images/videos.

[image]
[image]