Nova-UTD/navigator #381: Weather classification

danielv012, 2023-10-27:


22adubey, 2025-02-17:
**02/17/2025 Update:**
**Status:** Just picked up - found a dataset and trained a model [Link To Notebook](https://colab.research.google.com/drive/1L_jEQEhTMbd_B2jkXerX0yS_RNDbYVgL?usp=sharing)
**Next Steps:** Testing the model - first using grayscale images and then RGB images (from a car's POV) from Hail Bop or Google. If I've determined the model does not need further fine-tuning, I will begin creating a ROS node for weather classification.
**Projected Completion:** 1-2 weeks
**Update:** Found a dataset and trained a model.

22adubey, 2025-02-24:
### 02/24/2025 Update:
**Status:** Trained Mode and ran evaluation - noticed low accuracy for a few classifications. I'm going to try and find a new dataset and train a new model [Link To Training](https://colab.research.google.com/drive/1L_jEQEhTMbd_B2jkXerX0yS_RNDbYVgL?usp=sharing)   [Link to Eval](https://colab.research.google.com/drive/1Iik44okVFN_Ox3HwCb6lIj97gSa9WTUl?authuser=2#scrollTo=CM-of0wtd6Ql)

**Next Steps:** Finding a new dataset, cleaning it, and training a new model

**Projected Completion:** 2-3 weeks

**Update:** I trained my model and noticed poor performance in certain classifications. After fine-tuning, I saw some improvement, but a few categories still performed no better than random guessing. I suspect this is due to limitations in the quality or quantity of data for those classifications. To address this, I’m exploring new datasets, with BDD100K currently looking like a promising option.

22adubey, 2025-03-11:
### 03/10/2025 Update:
**Status:** Finished fine-tuning model and trained off the new dataset

**Next Steps:** Complete evaluation

**Projected Completion:** 2-3 weeks

**Update:** Finetuned and trained the model [Link to Kaggle notebook].(https://www.kaggle.com/code/aryamandub/weather-daytimeclassification/edit) Currently working on model evaluation

22adubey, 2025-03-24:
### 03/24/2025 Update:
**Status:** Evaluated model

**Next Steps:** Continue improving model accuracy

**Projected Completion:** 2-3 weeks

**Update:** Evaluated the model on the eval dataset [Link to Kaggle notebook](https://www.kaggle.com/code/aryamandub/weather-daytimeclassification/edit). Achieved 88% accuracy on the time of day classifier and about 60% accuracy on the weather classifier. I'm going to try and improve the weather classification this week.

22adubey, 2025-04-01:
03/31/2025 Update:
Status: Worked on improving Model

Next Steps: Continue improving model accuracy - For Rain and Snow

Projected Completion: 2-3 weeks

Update: [Link to Kaggle notebook](https://www.kaggle.com/code/aryamandub/weather-daytimeclassification/). So the time of day portion of the model looks good and the model is achieving 70%+ accuracy in detecting cloudy or clear weather types. However, the model performs poorly on rain and snow detection. I need to figure out how to improve rain and snow detection/

22adubey, 2025-04-21:
04/14/2025 Update:
Status: Worked on improving Model

Next Steps: Integrate ROS node?

Projected Completion: 2-3 weeks

Update: [Link to Kaggle notebook](https://www.kaggle.com/code/aryamandub/weather-daytimeclassification/). Continued working on Snow and rain detection. Perhaps it's time to start building the ROS node?