Nova-UTD/navigator #391: Refactor grid_summation_node

jruths, 2023-10-30:
- this node fast-forwards past grids based on the transform difference at the time points.  There could be some minor issues still with that. It also uses a method that converts the grid to an image and then rotates and translates, but there may be a more efficient way to do this.
- lots of hard coded values

pranavboyapati, 2025-03-07:
### 03/07/2025 Update:

- **Status:** Reviewed the existing node and identified all issues with it
- **Next Steps:** I will fix the first three issues detailed below: Clean up the imports, Replace hard-coded values with variables and formulas, and Fix the grid staleness checking function
- **Projected Completion:** Monday, March 17
- **Update:**  I reviewed all of the code in the existing grid summation node and ran it to see how well it works and what issues exist. I found the following issues with the existing node: 

     •  Clean up imports (lots of unnecessary imports)
     •	Remove hard coded values and replace with variables/formulas
     •	Fix the grid staleness checking function (Currently is 50%+ commented out)
     •	Finish the fast forwarding function (50%+ commented out so it doesn’t work properly). Remove if possible
     •	Improve efficiency of grid resizing functions (eliminate the type conversions)
     •	Finish the build cost map function (Currently doesn’t include all occupancy grids and is partially commented out)

     I will begin by fixing the first three issues in the list in the upcoming week, and then will move on to the others.

pranavboyapati, 2025-03-21:
### 03/21/2024 Update:

- **Status:** Finished cleaning up imports, replacing hard-coded values, and fixing the staleness checking function
- **Next Steps:** Finish the fast-forwarding function and grid resizing function
- **Projected Completion:** Monday, March 31
- **Update:**  I removed a lot of the unnecessary import statements, replaced hard-coded values with formulas, and fixed/cleaned up the grid staleness checking function. In the following week, I will finish and clean up the fast forwarding function, which is more than 50% commented out, and I will make the grid resizing function more efficient, as it is currently very slow.

pranavboyapati, 2025-03-29:
### 03/28/2025 Update:

- **Status:** Improved fast-forwarding and grid-resizing functions
- **Next Steps:** Finish the build-cost-map function
- **Projected Completion:** Monday, April 7
- **Update:**  I simplified the grid-resizing function by removing some of the type conversions that existed. I also fixed the fast-forwarding function by simplifying some of the logic and ensuring no values were hard-coded into this function. In the following week, I will finish/fix the build-cost-map function, which is the last part of the grid_summation_node that needs to be changed.

pranavboyapati, 2025-04-04:
### 04/04/2025 Update:

- **Status:** Finished making all the code changes and improvements
- **Next Steps:** Test out the node
- **Projected Completion:** Monday, April 14
- **Update:**  I fixed the build-cost-map function, so all the code changes and improvements have been completed. I need to test the updated node to ensure that it still works as expected. I will test using ROS bags and CARLA and am expecting to spend a week on this.

pranavboyapati, 2025-04-12:
### 04/12/2025 Update:

- **Status:** Finished testing and made a pull request
- **Update:**  This has been fully tested and it works, so I made a pull request. There are some issues coming up from the automated tests that run on pull requests, and I was unable to solve these so asked for help on that.

closed 2025-07-01 by jruths