Nova-UTD/navigator #456: Look into Github's large data storage for navigator model files

jruths, 2025-06-19:


pranavboyapati, 2025-06-25:
Added files with LFS. Look into cleaning up history to save more space

pranavboyapati, 2025-06-26:
New order of steps for reducing repository size:

1. Merge in all pull requests
2. Store files using Git LFS
3. Reduce the .git folder size using the following: https://web.archive.org/web/20190207210108/http://stevelorek.com/how-to-shrink-a-git-repository.html
4. Erase repository history (only if necessary after other steps)

pranavboyapati, 2025-07-08:
Steps Followed

1. Delete history following the instructions in ChatGPT chat titled “Delete Git History”
2. Remove LFS using git lfs uninstall
3. Follow the instructions at https://web.archive.org/web/20190207210108/http://stevelorek.com/how-to-shrink-a-git-repository.html
4. Run the following command: 
```
git -c gc.reflogExpire=0 -c gc.reflogExpireUnreachable=0 -c gc.rerereresolved=0 \
    -c gc.rerereunresolved=0 -c gc.pruneExpire=now gc
```

5. Run the following commands:  `git reflog expire --expire-unreachable=now --all git gc --prune=now `
6. Push changes to remote

**NOTE**: `git log --all —(pathname)` shows the tags referencing a file that no longer exists in latest version
**NOTE**: Following step 3 directly causes the remote and local repositories to diverge, so it is not possible to push changes to remote. Removing tags, keeping history, and trying step 3 causes the same problem.
**NOTE**: Removing tags and history cut the size down to 250MB
**NOTE**: Make sure to follow step 2 above no matter what action for size reduction is being taken

pranavboyapati, 2025-07-14:
- [ ] Re-open all PRs
- [ ] Make sure all branches can be merged into dev

closed 2025-07-14 by pranavboyapati